JoaoLopesF/SPFD5408 #3: Inverse colors

MaitreDede, 2015-12-08:
Hi,

When trying samples my color are inversed (red is cyan, black is white...)

In my app, I have to inverse all color bits to get the right colors :)

JoaoLopesF, 2016-01-25:
Hi,

Please modify the SPFD5408_Adafruit_TFTLCD line 307

before:

	// *** SPFD5408 change -- Begin
	// Not tested yet
	//writeRegister8(ILI9341_INVERTOFF, 0);
	//delay(500);

after:

	// *** SPFD5408 change -- Begin
	// Not tested yet
	writeRegister8(ILI9341_INVERTOFF, 0);
	//delay(500);

rohanraut92, 2016-05-24:
Im getting 2 half screens. Idk how to resolve it.

manniakk, 2016-08-22:
writeRegister8(ILI9341_INVERTON, 0);

this is commented
//writeRegister8(ILI9341_INVERTOFF, 0);

This fixed the color issue on my LCD

nopnop2002, 2017-05-04:
Hello manniakk.
Your TFT controller is NOT SPFD5408.
It's ILI9342.

https://forum.arduino.cc/index.php?topic=473885.0

Mark42XLII, 2017-09-12:
In my case, pure colors and secondary colors kinda work fine, but the other colors and black always are weird. Like, black looks like purple.

nopnop2002, 2017-09-12:
Hello manniakk.
Try this.
https://github.com/prenticedavid/MCUFRIEND_kbv/tree/master/examples/LCD_ID_readreg
You can see your tft driver.

closed 2018-09-26 by JoaoLopesF