JoaoLopesF/SPFD5408 #4: Touch Y coordinate is inverted

DominikPalo, 2015-12-17:
I'm using your library with a 2.4 TFT touchscreen bought on eBay. All works fine, except the touch functionality -  every `TSPoint` obtained using `TouchScreen.getPoint()` has inverted the Y-coordinate. It would by nice to have an option to set coordinate system for `TouchScreen` instance (for example as optional parameter of the `TouchScreen` constructor).

JoaoLopesF, 2016-01-25:
Hi,

In mine TFTs I need invert the coords.
Can be in your case it not necessary.

Please modify the SPFD5408_TouchScreen, line 159:

after:

//return TSPoint(x, y, z);

// *** SPFD5408 change -- Begin
// SPFD5408 change, because Y coordinate is inverted in this controller
return TSPoint(x, 1023 - y, z);
// -- End

before:

return TSPoint(x, y, z);

// *** SPFD5408 change -- Begin
// SPFD5408 change, because Y coordinate is inverted in this controller
//return TSPoint(x, 1023 - y, z);
// -- End

If this ok, I can create an #define for this

Please send me if this ok now.

Regards

Joao

hevergreen, 2017-03-05:
Half solved... my touch is now correct, but when i draw, up/down is left/right and vice-versa!!

detectingdragon77, 2017-08-09:
Thank You, the only working way ive found on the internet!!!!

mike-mustermann, 2018-01-04:
For me this work fine. I played with the TSPoint parameters. And BAM!
// *** SPFD5408 change -- Begin
   // SPFD5408 change, because Y coordinate is inverted in this controller
   return TSPoint(1100 - y, 1023 - x, 1023 - z);
   // -- End

closed 2018-09-26 by JoaoLopesF