JoaoLopesF/SPFD5408 #9: mirrored text

RafaEspG, 2016-04-26:
First, thanks for the code, and an apology for my bad english.

It work very fine (the touch screen and colors, works fine) but i have a problem when im using Arduino Mega 2560, the text is mirrored and coordinates of the drawings too, and i can´t fix this.

Can you help me, please?

nopnop2002, 2017-05-04:
Perhaps Your TFT controller is NOT SPFD5408.
It's ILI9342.

https://forum.arduino.cc/index.php?topic=473885.0

JoaoLopesF, 2017-05-08:
HI, in line 309 of SPFD5408_Adafruit_TFTLCD.cpp, try uncomment this lines:
        // *** SPFD5408 change -- Begin
	// Not tested yet
	//writeRegister8(ILI9341_INVERTOFF, 0);
	//delay(500);
Regards
Joao

wkyong, 2017-10-10:
I am using your library for my 2.8inch TFT screen from MCUfriend. I manage to get it to display the program using identifier ID 0x7575. However I find the text all mirrored. I had tried to add in the code 0x16, 0x08 to set the MY, MX and MV to 0, 0, 0 but failed. I am not able to find which code actually inverted the screen. Supplier of this screen claim they use a SPFD 5408 controller which lead me to this library. Please help me by providing some insight on which code I should change to rectify this problem.

wkyong, 2017-10-11:
I had my problem solved. Upon checking the lcd ID, it shows 4747 which link to HX8437D lcd controller. By using the correct library the TFT screen works flawlessly.

marcudanf, 2018-09-08:
@wkyong I have the same issue, can you please send a link to that library??

closed 2018-09-26 by JoaoLopesF