JoaoLopesF/SPFD5408 #15: touch inverted along the diagonal

haribukke, 2017-08-12:
I have referred issues 4 and 6 and mine is somewhat similar. my touch coordinates is inverted along the diagonal which moves from top left to bottom right.

The touch scenarios is as follows in paint:

1. I click on top left, the point is displayed correctly on the top left.
2. I click on bottom right, the point is displayed correctly on the bottom right. 
3. I click on top right, the point is displayed on the bottom left.
4. I click on bottom left, the point is displayed on top right.

It was initially fully inverted, when i uncommented line 159 of SPFD5408_TouchScreen.cpp and commented line 163. Please help me on how to fix this issue.

Mark42XLII, 2017-09-12:
Check if you TFT's pins are correct. I experienced something very similar. If in your shield the touchscreen pins (YP,YM,XP,XM) are in a different order or you wired your board wrongly, that happens. Try reverting the comment changes you did and use some male-female jumper wires to change the order of the touchscreen pins.

JoaoLopesF, 2017-09-18:
Hi,

You can try change  lines 159-164 of file SPFD5408_TouchScreen.cpp to
   return TSPoint(x, y, z);

   // *** SPFD5408 change -- Begin
   // SPFD5408 change, because Y coordinate is inverted in this controller
   //return TSPoint(x, 1023 - y, z);
   // -- End

playinmyblues, 2017-09-26:
This looks like the correct comment for my problem. For the calibration sketch on an Uno, the dots would appear top left and bottom right. I could select them and the program would continue. The numbers would be provided. Then after touching the screen to proceed, when I touched the screen and moved my finger, the line of dots would appear on another section.
Here is what was happening:
Draw a rectangle representing the touchscreen.
1) Start at the bottom left corner and drag to the right along the x axis. The line appeared at the top right and moved down.
2) Start at the top left corner and drag to the right along the x axis. The line appeared at the top left corner and moved down.
3) Start at the top left corner and drag down along the y axis. The line appear at the top left and moved right.
4) Start at the top right corner and drag down along the y axis. The line appeared at the bottom left and moved right.
I fumbled with this situation trying to get the correct changes in SPFD5408_TouchScreen.cpp but I was following the recommendations found in the "Touch Y coordinate is inverted". See the section about uncommenting and commenting lines as well as adding in a number.

To get a better handle on what was happening, I drew the rectangle as described above. Then I went to line 159 of the file. In the end, it amounts to uncommenting that line and switching the x and y variables:
return TSPoint(y, x, z);

That is what worked for my situation.
By the way, my TFT shield is probably a couple of years old as of 2017.

closed 2018-09-26 by JoaoLopesF