JoaoLopesF/SPFD5408 #16: 2 or more duplicated screen

txepkine, 2017-09-22:
Hi.
I'm sorry for my english.
I tested the examples and they work ok, but there is one problem. For example, in graphic test, I see 2 or 3 screen at the same time. When layout is vertical I see 2, when it's horitzontal I see 3. It happens the same in touchscreen test nd so. Any idea please? Thanks a lot.

txepkine, 2017-09-22:
I load an image to help what happens https://drive.google.com/open?id=0Byiryg5CAf2qWWRVWTZRc2d0dTA
Thanks a lot again

DesironnA, 2017-09-25:
Hey
I have this issue too. my arduino model is Uno R3. 
screenshot : 
http://s9.picofile.com/file/8307408100/20170925_160228.png

txepkine, 2017-09-25:
Hi DesironnA
Exactly the same for me. I use a Arduino one clone DXRDuino.
If I play with rotate parameter in the source code, it changes, but always with the same problem. The "void" space disappears but there are two screen.
I test width () and heigh() functions and it seems they detect a screen with the expected resolution... My knowledge is really low. No idea how to solve it....

DesironnA, 2017-09-26:
I thing out shields are intact. in other websites i see same problem and on answer..! somewhere, someone said this shield is a chines product with no library..!
maybe its chipset is ili9342 or SPFD5408. he tried both libraries but same result. :|

txepkine, 2017-09-26:
Hi again
Spdf5408_graphictest source code has 30 lines (68-96) to determine which chipset is used. These lines are commented, so they are not compiled  and  HX8347G LCD driver (0x9341) is used by defect (line 100). I uncomment these lines and I see that the chipset detected is in fact 0x9341 (as default).
So I think we are using the correct driver...... I don't know!!!!
I used another library, with the same results, but I forgot the library name...... I will find it if I can

DesironnA, 2017-09-26:
Hey again
I tried some other library. but same result.
but you can test it. maybe yours will be ok.
this library is different as my friend said. let me know your result :
http://s8.picofile.com/d/8307490826/3f1d019c-d02d-43d2-bfee-2c74708b085d/SPFD5408_master_1_.zip

txepkine, 2017-09-26:
I found it, https://github.com/brucetsao/eDisplay/tree/master/Arduino_Demo_UNO-Mega2560, but only works simpletest/simpletestfor uno/9341uno, but with the same results, two-three screens. A help image
https://drive.google.com/open?id=0Byiryg5CAf2qWGc5MlA2NVFmdTg
I think that the other examples doesn't work because are only for mega...

txepkine, 2017-09-26:
Hi DesironnA
I think that your suggested lib is exactly the same than Joao's one. Thanks anyway!

DesironnA, 2017-09-27:
The "SPFD5408-master" library is for 2.8 inch. not 2.4 inch..!
I'm thinking  about buy the other one. my eyes burned on reshearching about this baloney crap.
Our shields are OK. the RIGHT library is the problem..

txepkine, 2017-09-27:
Hi...
 I look at the readme file and it says:
"This library is the Adafruit TFT Libraries changed to works in TFT 2.4 shields with the SPFD5408 controller."
My tft screen is [http://img.dxcdn.com/productimages/sku_350121_2.jpg], as it says is 2,4 and mcufriend, and looks very similar (but not identical) than the showed by Joao here [https://www.youtube.com/watch?v=Q8WvEjXvAws]
I have been trying different things without exit, but it was expected as I'm not a good programmer. Probably I will buy another one too... I'm not able to solve this problem ....

PrinceCharles23, 2017-11-20:
hey there!
is there any news on the issue?
i tried a few different libraries and googled the heck out of it. i found a lot of people with the same issue but none of the suggested approaches (basically other libraries) worked

JoaoLopesF, 2017-11-20:
Hi,
I think this is another controller than the SPFD5408, have a lot of it in this chinesse 2.4 TFT.
Please see this site, to help you a identify the controller e correct lib for this.
http://misc.ws/2015/01/24/lcd-touch-screen-information/

Regards

Joao

DesironnA, 2017-11-28:
This issue is impossible to solve..

PrinceCharles23, 2017-12-06:
> Hi,
> I think this is another controller than the SPFD5408, have a lot of it in this chinesse 2.4 TFT.
> Please see this site, to help you a identify the controller e correct lib for this.
> http://misc.ws/2015/01/24/lcd-touch-screen-information/

been there, done that. id=0x00. tried the suggested library (and many others). it seems the display is plain broken...!?!

JoaoLopesF, 2017-12-07:
Hi, I think too its is broken :-(

I now working with Nextion display, fast design in PC app, connection only with 4 wires.
I suggest You try this display

Regards 

Joao

closed 2017-12-07 by JoaoLopesF