JoaoLopesF/SPFD5408 #17: MODELO COMPATIVEL?

paulofeju, 2017-10-28:
Ola João Lopes, comprei esse modulo de display TFT 2.8": 
https://www.ebay.com/itm/2-8-TFT-LCD-Shield-Touch-Panel-Module-Display-Transfer-PCB-Board-fr-Arduino-coi/322593066705?hash=item4b1c0b8ed1:g:h-QAAOSwEUVZaaIC
Estou tentando fazê-lo funcionar nessa placa Arduino MEGA 2560: https://www.ebay.com/itm/MEGA-2560-R3-ATMEGA16U2-ATMEGA2560-16AU-Board-USB-Cable-For-Arduino/272386431938?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649
Pesquisei bastante, tentei varias bibliotecas originais e alternativas,mas somente aparece a tela branca. Realizei todos os passos descritos por voce mas não deu certo. Não sei mais o que fazer...
Voce poderia me dar um caminho para fazer funcionar esse display na MEGA?
Desde ja agradeço demais por sua orientação.
Um grande abraço! Paulo

JoaoLopesF, 2017-11-20:
Oi, Paulo
Esta biblioteca é para TFT 2.4 com o controlador SPDF5402,
Por favor, acesse este site: http://misc.ws/2015/01/24/lcd-touch-screen-information/ 
Para Ajuda-lo a identificar o controlador e uma biblioteca para seu TFT.

Quer uma sugestão: Eu agora só estou trabalhando com Nextion, design rápido via app no PC, e conexão somente com 4 fios.

Abraços

João

Hi Paulo
This library is for TFT 2.4 with the SPDF5402 controller,
Please go to this site: http://misc.ws/2015/01/24/lcd-touch-screen-information/ To Help you identify the driver and library for your TFT.

Want a suggestion: I'm now only working with Nextion, fast design via app on PC, and only 4-wire connection.
Regards

Joao

closed 2017-11-20 by JoaoLopesF

paulofeju, 2017-11-21:
Muito obrigado João!!! Vou pesquisar sobre o Nextion.

LCD Touch Screen Information – misc.ws<http://misc.ws/2015/01/24/lcd-touch-screen-information/>
misc.ws
Most of the traffic I receive on this site is from people looking for additional information about the small, inexpensive touch screen LCDs commonly sold on eBay.

LCD Touch Screen Information – misc.ws<http://misc.ws/2015/01/24/lcd-touch-screen-information/>
misc.ws
Most of the traffic I receive on this site is from people looking for additional information about the small, inexpensive touch screen LCDs commonly sold on eBay.