JoaoLopesF/SPFD5408 #18: Compile fro Arduino Due

Nullfati, 2017-11-07:
I try compile the TFTPaint example for Arduino Due but have the errors.
My Arduino IDE version: 1.8.5
Arduino SAM Boards (32-bits ARM Cortex-M3): 1.6.11
Errors list:
In file included from C:\Users\Nullfati-note\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x/pins_arduino.h:20:0,

from C:\Program Files (x86)\Arduino\libraries\SPFD5408-master\SPFD5408_TouchScreen.cpp:6:

C:\Program Files (x86)\Arduino\libraries\SPFD5408-master\SPFD5408_TouchScreen.cpp: In member function 'TSPoint TouchScreen::getPoint()':

C:\Users\Nullfati-note\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x/variant.h:63:65: error: invalid conversion from 'Pio*' to 'uint8_t {aka unsigned char}' [-fpermissive]

 #define digitalPinToPort(P)        ( g_APinDescription[P].pPort )

C:\Program Files (x86)\Arduino\libraries\SPFD5408-master\SPFD5408_TouchScreen.cpp:61:21: note: in expansion of macro 'digitalPinToPort'

   uint8_t xp_port = digitalPinToPort(_xp);

C:\Users\Nullfati-note\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x/variant.h:63:65: error: invalid conversion from 'Pio*' to 'uint8_t {aka unsigned char}' [-fpermissive]

 #define digitalPinToPort(P)        ( g_APinDescription[P].pPort )

C:\Program Files (x86)\Arduino\libraries\SPFD5408-master\SPFD5408_TouchScreen.cpp:62:21: note: in expansion of macro 'digitalPinToPort'

   uint8_t yp_port = digitalPinToPort(_yp);

C:\Users\Nullfati-note\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x/variant.h:63:65: error: invalid conversion from 'Pio*' to 'uint8_t {aka unsigned char}' [-fpermissive]

 #define digitalPinToPort(P)        ( g_APinDescription[P].pPort )

C:\Program Files (x86)\Arduino\libraries\SPFD5408-master\SPFD5408_TouchScreen.cpp:63:21: note: in expansion of macro 'digitalPinToPort'

   uint8_t xm_port = digitalPinToPort(_xm);

C:\Users\Nullfati-note\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x/variant.h:63:65: error: invalid conversion from 'Pio*' to 'uint8_t {aka unsigned char}' [-fpermissive]

 #define digitalPinToPort(P)        ( g_APinDescription[P].pPort )

C:\Program Files (x86)\Arduino\libraries\SPFD5408-master\SPFD5408_TouchScreen.cpp:64:21: note: in expansion of macro 'digitalPinToPort'

   uint8_t ym_port = digitalPinToPort(_ym);

C:\Users\Nullfati-note\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x/variant.h:66:44: error: base operand of '->' is not a pointer

 #define portOutputRegister(port)   ( &(port->PIO_ODSR) )

C:\Program Files (x86)\Arduino\libraries\SPFD5408-master\SPFD5408_TouchScreen.cpp:77:4: note: in expansion of macro 'portOutputRegister'

   *portOutputRegister(yp_port) &= ~yp_pin;

C:\Users\Nullfati-note\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x/variant.h:66:44: error: base operand of '->' is not a pointer

 #define portOutputRegister(port)   ( &(port->PIO_ODSR) )

C:\Program Files (x86)\Arduino\libraries\SPFD5408-master\SPFD5408_TouchScreen.cpp:78:4: note: in expansion of macro 'portOutputRegister'

   *portOutputRegister(ym_port) &= ~ym_pin;

C:\Users\Nullfati-note\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x/variant.h:66:44: error: base operand of '->' is not a pointer

 #define portOutputRegister(port)   ( &(port->PIO_ODSR) )

C:\Program Files (x86)\Arduino\libraries\SPFD5408-master\SPFD5408_TouchScreen.cpp:86:4: note: in expansion of macro 'portOutputRegister'

   *portOutputRegister(xp_port) |= xp_pin;

C:\Users\Nullfati-note\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x/variant.h:66:44: error: base operand of '->' is not a pointer

 #define portOutputRegister(port)   ( &(port->PIO_ODSR) )

C:\Program Files (x86)\Arduino\libraries\SPFD5408-master\SPFD5408_TouchScreen.cpp:87:4: note: in expansion of macro 'portOutputRegister'

   *portOutputRegister(xm_port) &= ~xm_pin;

C:\Users\Nullfati-note\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x/variant.h:66:44: error: base operand of '->' is not a pointer

 #define portOutputRegister(port)   ( &(port->PIO_ODSR) )

C:\Program Files (x86)\Arduino\libraries\SPFD5408-master\SPFD5408_TouchScreen.cpp:102:5: note: in expansion of macro 'portOutputRegister'

    *portOutputRegister(xp_port) &= ~xp_pin;

C:\Users\Nullfati-note\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x/variant.h:66:44: error: base operand of '->' is not a pointer

 #define portOutputRegister(port)   ( &(port->PIO_ODSR) )

C:\Program Files (x86)\Arduino\libraries\SPFD5408-master\SPFD5408_TouchScreen.cpp:106:5: note: in expansion of macro 'portOutputRegister'

    *portOutputRegister(yp_port) |= yp_pin;

C:\Users\Nullfati-note\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x/variant.h:66:44: error: base operand of '->' is not a pointer

 #define portOutputRegister(port)   ( &(port->PIO_ODSR) )

C:\Program Files (x86)\Arduino\libraries\SPFD5408-master\SPFD5408_TouchScreen.cpp:125:5: note: in expansion of macro 'portOutputRegister'

    *portOutputRegister(xp_port) &= ~xp_pin;

C:\Users\Nullfati-note\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x/variant.h:66:44: error: base operand of '->' is not a pointer

 #define portOutputRegister(port)   ( &(port->PIO_ODSR) )

C:\Program Files (x86)\Arduino\libraries\SPFD5408-master\SPFD5408_TouchScreen.cpp:129:5: note: in expansion of macro 'portOutputRegister'

    *portOutputRegister(ym_port) |= ym_pin;

C:\Users\Nullfati-note\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x/variant.h:66:44: error: base operand of '->' is not a pointer

 #define portOutputRegister(port)   ( &(port->PIO_ODSR) )

C:\Program Files (x86)\Arduino\libraries\SPFD5408-master\SPFD5408_TouchScreen.cpp:133:5: note: in expansion of macro 'portOutputRegister'

    *portOutputRegister(yp_port) &= ~yp_pin;

exit status 1

JoaoLopesF, 2017-11-09:
Hi, sorry this library is not for Arduino DUE, only for AVR plataform

Regards

Joao

closed 2017-11-09 by JoaoLopesF

JoaoLopesF, 2017-11-09:
Hi, sorry this library is not for Arduino DUE, only for AVR platform

Regards

Joao

Nullfati, 2017-11-09:
After 2 days of tinkering i modified library for Due.